Gixy31/ESP32-PoolMaster #1: I can't compile code  error: aggregate 'tm timeinfo'

drivers1, 2021-03-16:
Hi,

I am trying to compile your code, but I have encountered several problems:
(Fixed) Can't find pump.h (result downloaded from logic code)
(Solved) Cannot find ADS1115 @ 1.0.0 from lib_deps (it generates an error that it cannot find the library ... but I have commented and downloaded it working)

Pending solution:
src \ Setup.cpp: 44: 4: error: aggregate 'tm timeinfo' has incomplete type and cannot be defined
  tm timeinfo;
     ^
Compiling .pio \ build \ serial_upload \ lib714 \ Easy Nextion Library \ readCustomCommands.cpp.o
*** [.pio \ build \ serial_upload \ src \ Setup.cpp.o] Error 1

You can help me?

Gixy31, 2021-03-16:
Hello,
Pump: https://github.com/Loic74650/PoolMaster
ADS1115: download it here: https://lygte-info.dk/project/ADS1115Library%20UK.html (not in GitHub)

Time and time is a long story... You have first to rename Time.h in _Time.h (Windows does'nt make difference between Time.h and time.h).

A new version is coming today, with a slightly modified timing of tasks in order to have a nice spreading of tasks along the cycle.

drivers1, 2021-03-16:
Hi, Thank you very much for your quick response.

I have tested on my macbook and I have the same problem (aggregate 'tm timeinfo' has incomplete type and cannot be defined).

Where I can find the file "Time.h" that I needs to change to "_Time.h"? Can test it on Mac or Windows. On both have the same problem.

Regards

Gixy31, 2021-03-17:
You should have a library called Time (paulstoffregen/Time @ 1.6). So in the folder of this library there's a header file called Time.h to rename in _Time.h. This folder is most probably in .pio/libdeps/serial_upload/Time, if you do not use the extra_libs folder in platformio.ini. The problem is that there is another header called time.h which is in the framework_arduinoespressif32 directory which must be included to declare the tm structure, but this Time.h file (which in turn includes time.h) put some garbage with the precompiler.

drivers1, 2021-03-17:
Solved!!! Thank you very much!!!

Other question. Could you share the file of the board "PoolMaster" you have designed? It would be good for me to know how are connected the sensors to esp32 and possibly to make one so that everything is more organized.

Regards!!

Gixy31, 2021-03-17:
OK. I'll do that later.
I close the issue.
Best Regards

closed 2021-03-17 by Gixy31

Gixy31, 2021-03-22:
pdf of the board added to repository.