Gixy31/ESP32-PoolMaster #2: Board

vaughnlfisher, 2021-03-29:
Where can i get my hands on the board

Gixy31, 2021-03-29:
Well, first by being polite, saying at least hello, please or something like that, which seems to be ignored by new generations... And second there is a pdf file called PoolMaster.pdf at the root of the directory with a 3D view and the schematic.

closed 2021-03-29 by Gixy31

vaughnlfisher, 2021-03-29:
> 
> 
> Well, first by being polite, saying at least hello, please or something like that, which seems to be ignored by new generations... And second there is a pdf file called PoolMaster.pdf at the root of the directory with a 3D view and the schematic.
My question was rude so i appoloise.

I am not of the new generation even though i wish i was sometimes.

The pdf is a good representation. where could I get the board made? Have you had it made?

Thanks,

Gixy31, 2021-03-29:
I usualy have my boards made by JLCPCB, and yes this one is made and operational.

vaughnlfisher, 2021-03-30:
Hi
Do you have a Gerber file for me please? JLPCB is asking for one?
Thanks again!

Gixy31, 2021-03-30:
zip file added. Enjoy.

vaughnlfisher, 2021-03-30:
Thanks 🙏 

Regards
Vaughn
Fisher

drivers1, 2021-03-30:
I'm interested too... Thanks very much!!!
One question. Are you order in JLPCB with option SMT Assembly??
Regards

Gixy31, 2021-03-31:
@drivers1 
No, I do it by myself. SMD are big enough to be soldered by hand, with a binocular to help.