Gixy31/ESP32-PoolMaster #6: Tank volume "problem"

vaughnlfisher, 2021-07-06:
Hi

Very good project my system is up and running, plus or minus. I have an issue potentially where the tank fill volume drops to zero in one movement. I dont understand why?

See attached picture of Grafan. Top panel show where over time ive set the tanks to 100% and randomly they drop to zero %

Any ideas?
[image]

Gixy31, 2021-07-06:
Hi,
Looking at pumps activities or ORP value, your system looks a bit strange. If you've modified the sofware, I can't help without seeing the code. I don't have this problem at all. The last version of my sofware has some modifications in the pump library and in the code to manage tank levels. May be you can try this new version.

vaughnlfisher, 2021-07-06:
Hi

I’ve not changed anything. Which specific files are you referring to? I have updated the settings etc but not code.

Thanks again

Gixy31, 2021-07-06:
Your pumps graph should look like this one. Did you set values in liters/min for the pH and chlorine pumps?
I encourage you to take the last version of the code and review your settings.
[image]

vaughnlfisher, 2021-07-06:
Thanks a million
I would expect the same graph as you. My graph suggests nothing is chinging in terms of the chlorine or ph pumps i.e. they stay off. If they stay off surely the tank should stay at 100%? I would expect a constant 100% on the tanks. Im testing outside of the pool environemt to make sure everything is good before i connect to the pool pipework.

vaughnlfisher, 2021-08-04:
Hi
I now have a pressure error at a 0.64 which I think is preventing any automatic dosing. My pump is not connected to the system, it’s independent. I have not modified any code.
[image]

vaughnlfisher, 2021-08-04:
thanks for your help
[image]

Hi changed it in the code, there is no where to set the medium threshold in node red. Still same problem?

Gixy31, 2021-08-04:
Normally there is one, called WPr Low in the settings dashboard (which is a more appropriate name).

vaughnlfisher, 2021-08-04:
Definitely nothing like that.
[image]

vaughnlfisher, 2021-08-04:
Hi

I don’t see anything like that?
Maybe I’ve done something wrong?

[image]

Sorry for being a bother

Gixy31, 2021-08-05:
Yes, this is the original Nodered dashboard. I modified it a lot since that. It is not so hard to add more fields such as Water Pressure  medthreshold, and you can then learn a lot on how Nodered works :-)