Gixy31/ESP32-PoolMaster #8: Gerber files

ippodrome, 2022-07-27:
Hi!
Thank you for this fork! 

I’m having some issues with the pcb when it’s was sent to the pcb-maker.

they sent me this:

[image]

i have no idea what this means or how to fix it. What should I do? 

Thank you! / viktor

Gixy31, 2022-07-27:
Hi,
Both are valid: the circles on the GKO layer ( Keep-out layer) are used to avoid traces routing too close from the holes. This is used only during PCB design. You can answer to your manufacturer to keep only drill holes and discard GKO layer.

closed 2022-07-27 by Gixy31

ippodrome, 2022-07-27:
Thank you very much for the quick reply! 👏🏼
Will tell them! 
😊