Gixy31/ESP32-PoolMaster #9: Software doesn't publish any MQTYT

vaughnlfisher, 2022-08-14:
Hi

I have recently lost all my original code! I have been trying to get the system back up and running with your latest code.
The nexion screen works and is reporting Realtime data. I can also upload OTA so the WIFI is working.

Problem is no information in being published via MQTT. I have a subscription and monitoring the Topic.

Attached terminal screen, nothing seems to happen after the Subscibe ack. qos:2

Could you assist

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[ 60 ] CPU frequency       : 240MHz
[ 61 ] CPU Cores           : 2
[ 61 ] Flash size          : 4MB
[ 61 ] Free RAM            : 282444 bytes
[ 72 ] Min heap            : 238072 bytes
[ 72 ] tskIDLE_PRIORITY    : 0
[ 72 ] confixMAX_PRIORITIES: 25
[ 72 ] configTICK_RATE_HZ  : 1000
[ 1396 ] Stored version: 51
[ 1397 ] Same version: 51 / 51. Loading settings from NVS
[ 1428 ] 51
[ 1428 ] 0, 0, 1, 0
[ 1429 ] 10, 10, 20, 8, 22, 20
[ 1429 ] 2700, 2700, 30000
[ 1429 ] 1800000, 1800000, 0, 0
[ 1429 ] 7.3,  720, 15.0, 15.0,  10,  18,  3.0, 3.496258, -2.011338, -876.430775, 2328.898500, 1.0, 0.0
[ 1440 ]  2700000,   0,   0,  18000,   0,   0,       0,       0, 28.00, 7.30,  720, 1.30
[ 1451 ]  60,  85,  20,  20, 1.5, 1.5
[ 1451 ] Config loaded
[ 1453 ] [WiFi] Connecting to WiFi...
[ 1631 ] [WiFi] Connection lost
[ 1716 ] [WiFi] Connected to: Fisher
[ 1716 ] [WiFi] IP address: 192.168.0.216
[ 1716 ] [WiFi] Hostname: PoolMaster
[ 1716 ] [WiFi] Connecting to MQTT...
[ 1760 ] Connected to MQTT, present session: 0
[ 1772 ] Subscribe ack., qos: 2
[ 2295 ] NTP configured
Sunday, August 14 2022 09:31:03
[ 2296 ] 2022/08/14 09:31:03
[ 2407 ] 1wire W devices: 1 device(s) found
[ 2408 ] 1wire A devices: 1 device(s) found
DS18B20_W: 28:a5:3f:8d:0c:00:00:bd
DS18B20_A: 28:aa:87:87:48:14:01:d5
[ 2530 ] Init done, starting loop tasks
[ 3631 ] [WiFi] Connecting to WiFi...
[ 3638 ] Disconnected from MQTT
[ 3661 ] [WiFi] Connection lost
[ 3802 ] [WiFi] Connected to: Fisher
[ 3803 ] [WiFi] IP address: 192.168.0.216
[ 3803 ] [WiFi] Hostname: PoolMaster
[ 3803 ] [WiFi] Connecting to MQTT...
[ 3840 ] Connected to MQTT, present session: 0
[ 3850 ] Subscribe ack., qos: 2
[ 5662 ] [WiFi] Connecting to WiFi...
[ 5668 ] Disconnected from MQTT
[ 5692 ] [WiFi] Connection lost
[ 5798 ] [WiFi] Connected to: Fisher
[ 5798 ] [WiFi] IP address: 192.168.0.216
[ 5798 ] [WiFi] Hostname: PoolMaster
[ 5798 ] [WiFi] Connecting to MQTT...
[ 5847 ] Connected to MQTT, present session: 0
[ 5855 ] Subscribe ack., qos: 2
[ 7692 ] [WiFi] Connecting to WiFi...
[ 7698 ] Disconnected from MQTT
[ 7722 ] [WiFi] Connection lost
[ 7827 ] [WiFi] Connected to: Fisher
[ 7827 ] [WiFi] IP address: 192.168.0.216
[ 7828 ] [WiFi] Hostname: PoolMaster
[ 7828 ] [WiFi] Connecting to MQTT...
[ 7884 ] Connected to MQTT, present session: 0
[ 7892 ] Subscribe ack., qos: 2
[ 9723 ] [WiFi] Connecting to WiFi...
[ 9729 ] Disconnected from MQTT
[ 9753 ] [WiFi] Connection lost
[ 10059 ] [WiFi] Connected to: Fisher
[ 10060 ] [WiFi] IP address: 192.168.0.216
[ 10060 ] [WiFi] Hostname: PoolMaster
[ 10060 ] [WiFi] Connecting to MQTT...
[ 10111 ] Connected to MQTT, present session: 0
[ 10117 ] Subscribe ack., qos: 2
[ 11753 ] [WiFi] Connecting to WiFi

Gixy31, 2022-08-14:
Hi,
Difficult to help from here, but according to the traces it seems you have a lot of WiFi disconnections, so MQTT disconnects as well.

vaughnlfisher, 2022-08-14:
Why would that be?

vaughnlfisher, 2022-08-14:
The WIFI is stable enough to do OTA.

Gixy31, 2022-08-14:
There are clearly WiFi events, but the traces are not in a logical order: should be 'connection lost' first, then 'connecting to wifi' and not the contrary.  Did you change the reconnect timers values (2000ms)? Another suggestion: try to update ESP32 Arduino core in PlatformIO.

closed 2022-09-03 by Gixy31