Gixy31/ESP32-PoolMaster #10: BOM obsolete

kifeo, 2023-05-31:
Hello,

Thanks for your work on this !
While getting the items from the BOM, I've found out that most of the links are not directing me to the correct items.

Would it be possible to update the BOM ?

Thanks !

kifeo, 2023-05-31:
It could be nice to be able to buy a package where you get some $$ percentage of it !

Gixy31, 2023-05-31:
Hum, remember that it is a DIY project, not a commercial one. It would be an endless work to continuously update the BOM from sellers like Amazon or AliExpress. Nevertheless, with the description of each component in the BOM, and with the pictures and schematics, I have no doubt that you can find everything to build the project. And then feel free to create a kit package.

closed 2023-05-31 by Gixy31