Gixy31/ESP32-PoolMaster #13: Erreur de compilation Olimex ESP32-EVB.

Fouggee, 2023-07-30:
Bonjour,

J'essai de compiler ESP32-PoolMaster sur un Olimex ESP32-EVB. La compilation échoue avec pluseurs erreurs concernant ADS1115 :

Library Manager: Installing ADS1115 @ 1.0.0
Warning! Could not find the package with 'ADS1115 @ 1.0.0' requirements for your system 'windows_amd64'

Alors plusieurs erreurs sortent du fichier src/Loops.cpp :

src/Loops.cpp:16:8: error: 'ADS1115Scanner' does not name a type; did you mean 'ADS1115'?
src/Loops.cpp:44:3: error: 'adc' was not declared in this scope
src/Loops.cpp:44:16: error: 'ADS1115_SPEED_16SPS' was not declared in this scope
etc ...

Peut-être avez-vous une idée?

Merci,
Jérôme.

Ci-dessous le résultat de la compilation dans le moniteur :

Processing serial_upload (platform: espressif32; board: esp32-evb; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-evb.html
PLATFORM: Espressif 32 (6.3.2) > OLIMEX ESP32-EVB
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Library Manager: Installing ADS1115 @ 1.0.0
Warning! Could not find the package with 'ADS1115 @ 1.0.0' requirements for your system 'windows_amd64'
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Arduino_DebugUtils @ 1.1.0
|-- Time @ 1.6.1
|-- PID @ 1.2.1
|-- I2Cdevlib-ADS1115 @ 1.0.0
|-- RunningMedian @ 0.2.1
|-- ArduinoJson @ 6.21.3
|-- Easy Nextion Library @ 1.0.6
|-- DallasTemperature @ 3.11.0
|-- AsyncTCP @ 1.1.1
|-- AsyncMqttClient @ 0.8.2
|-- OneWire @ 2.3.7
|-- Wire @ 2.0.0
|-- Pump-master
|-- Preferences @ 2.0.0
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- Update @ 2.0.0
|-- I2Cdevlib-Core @ 1.0.1
Building in debug mode
Compiling .pio\build\serial_upload\src\Loops.cpp.o
Compiling .pio\build\serial_upload\src\Nextion.cpp.o
Compiling .pio\build\serial_upload\src\PoolMaster.cpp.o
Compiling .pio\build\serial_upload\src\PoolServer.cpp.o
src/Loops.cpp:16:8: error: 'ADS1115Scanner' does not name a type; did you mean 'ADS1115'?
 static ADS1115Scanner adc(0x48);  // Address 0x48 is the default
        ^~~~~~~~~~~~~~
        ADS1115
src/Loops.cpp: In function 'void AnalogInit()':
src/Loops.cpp:44:3: error: 'adc' was not declared in this scope
   adc.setSpeed(ADS1115_SPEED_16SPS);
   ^~~
src/Loops.cpp:44:16: error: 'ADS1115_SPEED_16SPS' was not declared in this scope
   adc.setSpeed(ADS1115_SPEED_16SPS);
                ^~~~~~~~~~~~~~~~~~~
src/Loops.cpp:44:16: note: suggested alternative: 'ADS1115_RATE_16'
   adc.setSpeed(ADS1115_SPEED_16SPS);
                ^~~~~~~~~~~~~~~~~~~
                ADS1115_RATE_16
src/Loops.cpp:45:18: error: 'ADS1115_CHANNEL0' was not declared in this scope
   adc.addChannel(ADS1115_CHANNEL0, ADS1115_RANGE_6144);
                  ^~~~~~~~~~~~~~~~
src/Loops.cpp:45:18: note: suggested alternative: 'ADS1115_RATE_8'
   adc.addChannel(ADS1115_CHANNEL0, ADS1115_RANGE_6144);
                  ^~~~~~~~~~~~~~~~
                  ADS1115_RATE_8
src/Loops.cpp:45:36: error: 'ADS1115_RANGE_6144' was not declared in this scope
   adc.addChannel(ADS1115_CHANNEL0, ADS1115_RANGE_6144);
                                    ^~~~~~~~~~~~~~~~~~
src/Loops.cpp:45:36: note: suggested alternative: 'ADS1115_RATE_64'
   adc.addChannel(ADS1115_CHANNEL0, ADS1115_RANGE_6144);
                                    ^~~~~~~~~~~~~~~~~~
                                    ADS1115_RATE_64
src/Loops.cpp:46:18: error: 'ADS1115_CHANNEL1' was not declared in this scope
   adc.addChannel(ADS1115_CHANNEL1, ADS1115_RANGE_6144);
                  ^~~~~~~~~~~~~~~~
src/Loops.cpp:46:18: note: suggested alternative: 'ADS1115_RATE_16'
   adc.addChannel(ADS1115_CHANNEL1, ADS1115_RANGE_6144);
                  ^~~~~~~~~~~~~~~~
                  ADS1115_RATE_16
src/Loops.cpp:47:18: error: 'ADS1115_CHANNEL2' was not declared in this scope
   adc.addChannel(ADS1115_CHANNEL2, ADS1115_RANGE_6144);
                  ^~~~~~~~~~~~~~~~
src/Loops.cpp:47:18: note: suggested alternative: 'ADS1115_RATE_32'
   adc.addChannel(ADS1115_CHANNEL2, ADS1115_RANGE_6144);
                  ^~~~~~~~~~~~~~~~
                  ADS1115_RATE_32
src/Loops.cpp: In function 'void AnalogPoll(void*)':
src/Loops.cpp:67:3: error: 'adc' was not declared in this scope
   adc.start();
   ^~~
*** [.pio\build\serial_upload\src\Loops.cpp.o] Error 1
================================================= [FAILED] Took 20.69 seconds =================================================

Environment    Status    Duration
-------------  --------  ------------
serial_upload  FAILED    00:00:20.688
============================================ 1 failed, 0 succeeded in 00:00:20.688 ============================================ 

 *  The terminal process "C:\Users\xxxxxxx\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

Gixy31, 2023-07-31:
Bonjour,
Les messages sont pourtant clairs : vous n’avez pas installé la librairie ADS1115. Celle-ci n’est pas dans le registre général de PlatformIO et doit être installée manuellement. Le source est ici : https://lygte-info.dk/project/ADS1115Library%20UK.html
L'emplacement où vous installez la librairie doit figurer dans le fichier platformio.ini du projet (symlink).

closed 2023-07-31 by Gixy31

Fouggee, 2023-07-31:
Bonsoir,

Merci pour la mise à jour. 
Avec le Symlink la compilation fonctionne. 
Pour les nouveaux sur platformio il serait bien de l'ajouter dans le readme ou directement en commentaire dans le fichier platformio.ini.

Cordialement,
Jérôme

Gixy31, 2023-07-31:
Commentaire ajouté dans platformio.ini