Gixy31/ESP32-PoolMaster #14: CAD/Eagle File

aaronweaver, 2024-02-14:
Thanks for putting the project on github and supplying the code and documentation. Would it be possible to also upload the eagle/CAD file as well? I wanted to make a couple slight modifications on the board. Thanks!

Gixy31, 2024-02-14:
Hi,
I don’t use Eagle but Altium. Let me know if you can import Altium files and I will upload them.

Envoyé à partir de Courrier pour Windows

De : Aaron Weaver
Envoyé le :mercredi 14 février 2024 17:49
À : Gixy31/ESP32-PoolMaster
Cc : Subscribed
Objet :[Gixy31/ESP32-PoolMaster] CAD/Eagle File (Issue #14)

Thanks for putting the project on github and supplying the code and documentation. Would it be possible to also upload the eagle/CAD file as well? I wanted to make a couple slight modifications on the board. Thanks!

aaronweaver, 2024-02-14:
Yes I can import Altium files. Thanks again!

Gixy31, 2024-02-14:
Files uploaded, enjoy!

Envoyé à partir de Courrier pour Windows

De : Aaron Weaver
Envoyé le :mercredi 14 février 2024 19:07
À : Gixy31/ESP32-PoolMaster
Cc : Gixy31; Comment
Objet :Re: [Gixy31/ESP32-PoolMaster] CAD/Eagle File (Issue #14)

Yes I can import Altium files. Thanks again!

Gixy31, 2024-02-21:
Request done

closed 2024-02-21 by Gixy31