Gixy31/ESP32-PoolMaster #15: Bom for the card does not exist.

jacquesj1, 2024-03-19:
Hello,
The gerber files exist but, is it possible to have the bom file associated ?

Thanks

jacquesj1, 2024-03-19:
> BOM added.

Bonjour,
Super merci!
Je trouve votre projet très intéressant.
J'ai d'autres questions :
Avez-vous un schéma de connexion entre votre carte PoolMaster, les relais,
les pompes, .... ?
Comment intégrez- vous les chambres d'analyse ?

Encore bravo pour votre projet !
Cordialement,
Jérôme JACQUES

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sans
virus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Gixy31, 2024-03-19:
Bonjour,
Pas de schéma global. Vous avez des photos ici : https://forum.arduino.cc/t/poolmaster-gestion-et-domotisation-de-ma-piscine/563058/350?u=gixy31 
Je vous conseille de consulter le topic complet, certes plus de 1000 posts, mais vous devriez trouver réponses à beaucoup de question déjà posées sur les différentes versions du projet, dont je ne suis pas le concepteur initial comme indiqué en introduction.
Les sondes sont dans des chambres d'analyse en dérivation sur le circuit de filtration.
Les relais de la pompe de filtration, du surpresseur du robot et du projecteur activent des commutateurs dans l'armoire d'origine de la piscine, ce qui fait que les relais de PoolMaster ne coupent pas des courants forts mais seulement les bobines des commutateurs, elles-mêmes munies de snubbers pour éviter les parasites à la coupure. Les pompes peristaltiques sont directement reliées aux relais de PoolMaster car de faible puissance.

jacquesj1, 2024-03-20:
MERCI BEAUCOUP !

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sans
virus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>